coolsnowwolf/luci #12: 主题显示不一样

nature2608, 2019-10-09:
同一个ipk，我安装在您的源码编译的固件上会显示的很简单，有些地方也不正常。
[image]
[image]
[image]
而安装在openwrt官方的固件上则是正常的。
[image]
[image]
[image]

用的是官方带的rosy主题。我如果切换主题的话也是不正常。

outemancut, 2019-10-15:
让这个主题的开发者来做兼容就是了 默认的主题虽然没有你们的花里胡哨 但是是兼容跟性能最强的 这种问题要么自己做解决要么用默认主题！

nature2608, 2019-10-15:
@outemancut 兼容性的确是，用官方的luci很奇怪

closed 2020-03-02 by coolsnowwolf