coolsnowwolf/luci #34: b753998是否改错了

BoBppy, 2020-07-09:
https://github.com/coolsnowwolf/luci/commit/b753998006cad2c3751ac801d7164329c5224b8d

cpuinfo在第694行，而修改的在693行
表现为R2S架构仍显示?x4

[image]

@dukh1216 提供图片

BoBppy, 2020-07-09:
还有请问R2S能显示温度吗

klever1988, 2020-08-05:
cd lede/
wget https://github.com/klever1988/lede/blob/master/enable_autocore.diff
git apply enable_autocore.diff

[image]

dukh1216, 2020-08-06:
> cd lede /
> wget https://github.com/klever1988/lede/blob/master/enable_autocore.diff
> git apply enable_autocore.diff
> 
> [image]

git apply enable_autocore.diff一行，提示未能识别的输入，请问是直接在lede目录下输入吗？

klever1988, 2020-08-06:
> git apply enable_autocore.diff一行，提示未能识别的输入，请问是直接在lede目录下输入吗？
对 不过需要先 ./scripts/feeds install -a

dukh1216, 2020-08-06:
> > git apply enable_autocore.diff一行，提示未能识别的输入，请问是直接在lede目录下输入吗？
> 
> 对 不过需要先 ./scripts/feeds install -a

还是这样
[image]

klever1988, 2020-08-06:
> > > git apply enable_autocore.diff一行，提示未能识别的输入，请问是直接在lede目录下输入吗？
> > 
> > 
> > 对 不过需要先 ./scripts/feeds install -a
> 
> 还是这样
> [image]

抱歉给错diff的地址了，应该是
https://github.com/klever1988/lede/raw/master/enable_autocore.diff

dukh1216, 2020-08-06:
> > > > git apply enable_autocore.diff一行，提示未能识别的输入，请问是直接在lede目录下输入吗？
> > > 
> > > 
> > > 对 不过需要先 ./scripts/feeds install -a
> > 
> > 
> > 还是这样
> > [image]
> 
> 抱歉给错diff的地址了，应该是
> https://github.com/klever1988/lede/raw/master/enable_autocore.diff

大佬，重新编译了一次，还是不行(°ー°〃)

klever1988, 2020-08-06:
> 大佬，重新编译了一次，还是不行(°ー°〃)

Extra packages ->  autocore 要选中编译

dukh1216, 2020-08-06:
> > 大佬，重新编译了一次，还是不行(°ー°〃)
> 
> Extra packages -> autocore 要选中编译

可以了，感谢大佬耐心指点O(∩_∩)O

PR #37 (2020-08-10): luci: fix cpuinfo display on ARMv8 machine
fix #34
closed 2020-08-10 by coolsnowwolf