coolsnowwolf/luci #127: luci-app-filetransfer 在 imageBuilder 打包的时候报错 sed: can't read /usr/lib/lua/luci/dispatcher.lua: No such file or directory

zhangguanzhang, 2022-06-06:
```
$ make image PACKAGES="$PACKAGES ${extra_pkgs}" FILES="files"
...
...
Configuring luci-i18n-upnp-zh-cn.
Configuring kmod-usb-net-sr9700.
Configuring coreutils-cat.
Configuring kmod-dummy.
Configuring kmod-fs-squashfs.
Configuring kmod-net-prism54.
Configuring luci-app-docker.
Configuring iperf3.
Configuring tmate.
Configuring kmod-p54-usb.
Configuring luci-i18n-firewall-zh-cn.
Configuring luci-i18n-docker-zh-cn.
Configuring ppp-mod-pppoe.
Configuring luci-i18n-nlbwmon-zh-cn.
Configuring libnl-genl200.
Configuring ipvsadm.
Configuring luci-proto-ncm.
Configuring kmod-net-rtl8192su.

Finalizing root filesystem...
/root/op/openwrt-imagebuilder-rockchip-armv8.Linux-x86_64/staging_dir/host/bin/sed: can't read /usr/lib/lua/luci/dispatcher.lua: No such file or directory
/root/op/openwrt-imagebuilder-rockchip-armv8.Linux-x86_64/staging_dir/host/bin/sed: can't read /usr/lib/lua/luci/dispatcher.lua: No such file or directory
Enabling AdGuardHome
Enabling aliyundrive-webdav
Enabling aria2
Enabling arpbind
Enabling autoreboot
Enabling boot
Enabling cgroupfs-mount
...
...
```
然后我在文件 `package/lean/luci-app-filetransfer/root/etc/uci-defaults/luci-filetransfer` 里hack加了句
```
[ ! -f /etc/openwrt_release ] && exit 0
```
就没这个错误了，应该有 IPKG_INSTROOT 之类的 chroot 里去执行才对，我这种hack应该是不标准的

Beginner-Go, 2022-06-06:
您pr下

zhangguanzhang, 2022-06-06:
我对op的chroot部分不懂，不知道如何修复是符合要求的

zhangguanzhang, 2022-06-06:
> 您pr下

我对op的chroot部分不懂，不知道如何修复是符合要求的

coolsnowwolf, 2026-09-06:
当前 `luci-app-filetransfer` 已不再包含报告中提到的 `root/etc/uci-defaults/luci-filetransfer`，也没有任何针对 `/usr/lib/lua/luci/dispatcher.lua` 的 `sed` 操作；现行包仅安装 LuCI 页面、控制器和 ACL。该 imageBuilder 报错来源已移除，`unblockmusic` 的独立 init 报错不属于本 issue。

closed 2026-09-06 by coolsnowwolf