coolsnowwolf/luci #131: 最新的turboacc停止dns缓存后，dnsmasq的dns转发不能自动删除和取消忽略解析文件！

zxlhhyccc, 2022-06-19:
去掉https://github.com/coolsnowwolf/luci/blob/master/applications/luci-app-turboacc/root/etc/init.d/turboacc#L221， 这行代码能自动删除dnsmasq里的dns转发以及取消忽略解析文件。

WYC-2020, 2022-06-21:
这玩意直接去掉tur对dns的操作，基本没用，现在随便哪个梯子都有这个功能了，这个功能去掉，我自己仓库去掉，我看隔壁的也是去掉了，这样就不会有你这个问题了

zxlhhyccc, 2022-06-21:
> 这玩意直接去掉tur对dns的操作，基本没用，现在随便哪个梯子都有这个功能了，这个功能去掉，我自己仓库去掉，我看隔壁的也是去掉了，这样就不会有你这个问题了

大佬，能提供你的仓库吗？

windybird, 2022-07-03:
#144  看看这个应该就可以了

ntlf9t, 2022-08-04:
我自己用的魔改版，试试
[turboacc.txt](https://github.com/coolsnowwolf/luci/files/9260025/turboacc.txt)

coolsnowwolf, 2026-09-06:
当前 `luci-app-turboacc/root/etc/init.d/turboacc` 已不再包含旧版 `change_dns()/revert_dns()`、`/var/run/dnscache/.dns-changed`、`127.0.0.1#5333` 转发或 resolvfile 忽略逻辑；报告所指的代码已移除，因此该问题在现行版本中不存在。

closed 2026-09-06 by coolsnowwolf