coolsnowwolf/luci #132: Turbo ACC luci里无法修改状态

tone88, 2022-06-20:
[image]
取消勾选-保存&应用 ，状态仍然是运行，见上图，即使重启路由也还是这状态。

zxlhhyccc, 2022-06-20:
是的，最新版本我也是这个情况。。。

leochien0102, 2022-06-20:
+1，一直运行中

zxlhhyccc, 2022-06-20:
> 一样的问题，暂时修改feeds.conf，拉取指定tag（这次变更前）的luci源，先用着，等修复了再改回来 [image]

经实际测试，turboacc的脚本实际未调用shortcut-fe脚本，如果ssh进去操作:/etc/init.d/turboacc stop却可以调用shortcut-fe脚本，同时，shortcut-fe脚本需要在stop里添加：shortcut-fe才能正常

xiangfeidexiaohuo, 2022-06-21:
> > 一样的问题，暂时修改feeds.conf，拉取指定tag（这次变更前）的luci源，先用着，等修复了再改回来 [image]
> 
> 经实际测试，turboacc的脚本实际未调用shortcut-fe脚本，如果ssh进去操作:/etc/init.d/turboacc stop却可以调用shortcut-fe脚本，同时，shortcut-fe脚本需要在stop里添加：shortcut-fe才能正常 因此，可以暂时按照turboacc脚本的上文第26行： https://github.com/coolsnowwolf/luci/blob/master/applications/luci-app-turboacc/root/etc/init.d/turboacc#L26 的意思和shortcut-fe脚本的stop部分，如果运行shortcut-fe，应该检测运行的模块是shortcut_fe_cm.ko，所以将下文的第334行： https://github.com/coolsnowwolf/luci/blob/master/applications/luci-app-turboacc/root/etc/init.d/turboacc#L334 修改为：lsmod | grep -q "shortcut_fe_cm"; then 即可正常启用和停止。 上述修改不知道符合不符合要求，请大佬指正！但在未修复前，作为暂时性的使用。。。。

这么改确实可以关停了，看提交确实也是改动了这里了。
[image]

WYC-2020, 2022-06-21:
> 这么改确实可以关停了，看提交确实也是改动了这里了。 [image]

通过模块判断没问题
[image]
就应该改回来

WYC-2020, 2022-06-21:
问题不单单只有这一个 你试下关闭然后重启路由器看下 你在看下那个状态哈哈

Aes64X, 2022-06-21:
@AmadeusGhost  大佬，判断炸了 - -
相关：https://github.com/coolsnowwolf/lede/issues/9628

WYC-2020, 2022-06-21:
我已经修好

WYC-2020, 2022-06-22:
> > 我已经修好
> 
> **大佬，你修的在哪里啊？请问大佬能分享吗？**

[image]
和l大的有区别，自己修改哈，还有如果恢复了防火墙那个全锥形控制还需要吧原来init删除的那个脚本回复才行，要不然那边勾选了这边会出现没勾选

zxlhhyccc, 2022-06-22:
> > > 我已经修好
> > 
> > 
> > **大佬，你修的在哪里啊？请问大佬能分享吗？**
> 
> [image] 和l大的有区别，自己修改哈，还有如果恢复了防火墙那个全锥形控制还需要吧原来init删除的那个脚本回复才行，要不然那边勾选了这边会出现没勾选

请问是哪个脚本？是turboacc_fss这个脚本吗？

xiangfeidexiaohuo, 2022-06-22:
> > > > 我已经修好
> > > 
> > > 
> > > **大佬，你修的在哪里啊？请问大佬能分享吗？**
> > 
> > 
> > [image] 和l大的有区别，自己修改哈，还有如果恢复了防火墙那个全锥形控制还需要吧原来init删除的那个脚本回复才行，要不然那边勾选了这边会出现没勾选
> 
> 我改成这样子也是没问题的，目前貌似正常，但认为你这个改动更合理，另外我没恢复那个turboacc_fss脚本，不过没测试防火墙的全锥形控制： [image]

turboacc_fss这个文件还是要加，再就是这
[image]

不加的话，在防火墙里控制全锥形的开关，turboacc的全锥形的开关不会跟随变动。

zxlhhyccc, 2022-06-22:
> turboacc_fss这个文件还是要加，再就是这 [image]
> 
> 不加的话，在防火墙里控制全锥形的开关，turboacc的全锥形的开关不会跟随变动。

嗯，我测试了确实要加。。。。

zxlhhyccc, 2022-06-22:
> > > 我已经修好
> > 
> > 
> > **大佬，你修的在哪里啊？请问大佬能分享吗？**
> 
> [image] 和l大的有区别，自己修改哈，还有如果恢复了防火墙那个全锥形控制还需要吧原来init删除的那个脚本回复才行，要不然那边勾选了这边会出现没勾选

大佬，根据你修改的，全新编译的固件刷进去启动后shortcut-fe仍会自动运行，按照我上面我自己改的倒是正常。。。

WYC-2020, 2022-06-22:
我截图没截完，等会发给你

WYC-2020, 2022-06-23:
[luci-app-turboacc.tar.gz](https://github.com/coolsnowwolf/luci/files/8963017/luci-app-turboacc.tar.gz)
这是我的，去掉了dns的，如果不不想去掉那就按我这个改你相同部分就行
@zxlhhyccc

zxlhhyccc, 2022-06-23:
> [luci-app-turboacc.tar.gz](https://github.com/coolsnowwolf/luci/files/8963017/luci-app-turboacc.tar.gz) 这是我的，去掉了dns的，如果不不想去掉那就按我这个改你相同部分就行 @zxlhhyccc

收到，谢谢大佬！

zxlhhyccc, 2022-06-23:
> [luci-app-turboacc.tar.gz](https://github.com/coolsnowwolf/luci/files/8963017/luci-app-turboacc.tar.gz) 这是我的，去掉了dns的，如果不不想去掉那就按我这个改你相同部分就行 @zxlhhyccc

大佬，按照你发的改了，但是全新编译后刷入路由启动后仍是会自动运行。。。。

tone88, 2022-06-23:
能修复的大佬提个PR呗

xiangfeidexiaohuo, 2022-06-23:
> > [luci-app-turboacc.tar.gz](https://github.com/coolsnowwolf/luci/files/8963017/luci-app-turboacc.tar.gz) 这是我的，去掉了dns的，如果不不想去掉那就按我这个改你相同部分就行 @zxlhhyccc
> 
> 大佬，按照你发的改了，但是全新编译后刷入路由启动后仍是会自动运行。。。。

确实有这个现象，刷入后确实会自动运行，重启倒是不会了

WYC-2020, 2022-06-23:
不可能完全用我的那个绝对不会，你那个要自动运行是用的l大的，他那个配置文件设置就是启动

zxlhhyccc, 2022-06-23:
全新固件自动启动问题、系统日志“out of range”提示问题和关闭dns缓存后不会自动解除dnsmasq的dns转发相关问题已经全部搞定，明天我提pr或者做一个补丁发上来。。。。

WYC-2020, 2022-06-24:
> > 不可能完全用我的那个绝对不会，你那个要自动运行是用的l大的，他那个配置文件设置就是启动
> 
> 大佬 我用你的测试 确实会这样，我是格式化测试，全新刷入固件。刷入固件之前，我格式化了系统，当前sfe是关闭的。然后刷入固件，刷完开机，sfe确实自动启动了，也有sh: out of range这个日志。

有时间看下

zxlhhyccc, 2022-06-24:
大家拿去用吧，替换etc/init.d/目录下的同名文件就行（注意权限），此脚本解决了：全新固件自动启动问题、系统日志“out of range”提示问题和关闭dns缓存后不会自动解除dnsmasq的dns转发相关问题。另外，全锥型控制请自行添加那个脚本。
[turboacc.zip](https://github.com/coolsnowwolf/luci/files/8973389/turboacc.zip)

xiangfeidexiaohuo, 2022-06-24:
> **大家拿去用吧，替换etc/init.d/目录下的同名文件就行（注意权限），此脚本解决了：全新固件自动启动问题、系统日志“out of range”提示问题和关闭dns缓存后不会自动解除dnsmasq的dns转发相关问题。另外，全锥型控制请自行添加那个脚本。——不足之处请大佬们指正，烦请各位大佬不要喷我！谢谢啊！** [turboacc.zip](https://github.com/coolsnowwolf/luci/files/8973389/turboacc.zip)

测试了下确实没啥问题了，我整个完整的上来吧
[luci-app-turboacc.zip](https://github.com/coolsnowwolf/luci/files/8973830/luci-app-turboacc.zip)
。

WYC-2020, 2022-06-25:
> > 不可能完全用我的那个绝对不会，你那个要自动运行是用的l大的，他那个配置文件设置就是启动
> 
> 大佬 我用你的测试 确实会这样，我是格式化测试，全新刷入固件。刷入固件之前，我格式化了系统，当前sfe是关闭的。然后刷入固件，刷完开机，sfe确实自动启动了，也有sh: out of range这个日志。

刚看了大概知道什么原因了，考虑掉了一种情况，tru没有开启，但是sfe的服务是启用状态，在start这里的时候不会禁用sfe,所以导致的，有两种方法 一种就你那个方案，在tru启动的时候根据类型禁用或者启用，其实还有一种我觉得稍微更好的，就是sfe服务默认安装的时候就成禁用状态，这个好处可以减少系统加载不必要的模块，搞定

zxlhhyccc, 2022-06-25:
> 刚看了大概知道什么原因了，考虑掉了一种情况，tru没有开启，但是sfe的服务是启用状态，在start这里的时候不会禁用sfe,所以导致的，有两种方法 一种就你那个方案，在tru启动的时候根据类型禁用或者启用，其实还有一种我觉得稍微更好的，就是sfe服务默认安装的时候就成禁用状态，这个好处可以减少系统加载不必要的模块，搞定

大佬，能提供你的方案吗？我认为你的方案更好！谢谢啦！

xiangfeidexiaohuo, 2022-06-25:
> 刚看了大概知道什么原因了，考虑掉了一种情况，tru没有开启，但是sfe的服务是启用状态，在start这里的时候不会禁用sfe,所以导致的，有两种方法 一种就你那个方案，在tru启动的时候根据类型禁用或者启用，其实还有一种我觉得稍微更好的，就是sfe服务默认安装的时候就成禁用状态，这个好处可以减少系统加载不必要的模块，搞定

从理想状态来说，大佬这种确实好一些，哈哈，期待大佬的方案。并且还能去掉dns，这个dns服务，好多人不喜欢这里的。

tone88, 2022-06-25:
> 刚看了大概知道什么原因了，考虑掉了一种情况，tru没有开启，但是sfe的服务是启用状态，在start这里的时候不会禁用sfe,所以导致的，有两种方法 一种就你那个方案，在tru启动的时候根据类型禁用或者启用，其实还有一种我觉得稍微更好的，就是sfe服务默认安装的时候就成禁用状态，这个好处可以减少系统加载不必要的模块，搞定

这个问题讨论好几天总算有最优方案了，期待大佬的PR，大佬辛苦了！

closed 2022-06-25 by tone88

zxlhhyccc, 2022-06-25:
> 刚看了大概知道什么原因了，考虑掉了一种情况，tru没有开启，但是sfe的服务是启用状态，在start这里的时候不会禁用sfe,所以导致的，有两种方法 一种就你那个方案，在tru启动的时候根据类型禁用或者启用，其实还有一种我觉得稍微更好的，就是sfe服务默认安装的时候就成禁用状态，这个好处可以减少系统加载不必要的模块，搞定

我真的是小白一枚啊，大佬你说的意思我一下没反应过来搞了一晚上，才发现你的方案最简单了，就是将sfe脚本的start=99删除或注释掉，sfe服务就禁用状态了，你这个方案我认为确实是最优化。。。。对于系统日志“out of range”提示问题和关闭dns缓存后不会自动解除dnsmasq的dns转发相关问题比照我上面发的脚本修改对应地方即可。。。。