coolsnowwolf/luci #150: Windows自带VPN客户端无法连接IPsec服务器

lucky-beans, 2022-08-13:
除了Windows其它都能正常用
Sat Aug 13 12:10:33 2022 daemon.info : 08[NET] received packet: from 192.168.50.189[500] to 192.168.50.1[500] (408 bytes)
Sat Aug 13 12:10:33 2022 daemon.info : 08[ENC] parsed ID_PROT request 0 [ SA V V V V V V V V ]
Sat Aug 13 12:10:33 2022 daemon.info : 08[ENC] received unknown vendor ID: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
Sat Aug 13 12:10:33 2022 daemon.info : 08[IKE] received MS NT5 ISAKMPOAKLEY vendor ID
Sat Aug 13 12:10:33 2022 daemon.info : 08[IKE] received NAT-T (RFC 3947) vendor ID
Sat Aug 13 12:10:33 2022 daemon.info : 08[IKE] received draft-ietf-ipsec-nat-t-ike-02\n vendor ID
Sat Aug 13 12:10:33 2022 daemon.info : 08[IKE] received FRAGMENTATION vendor ID
Sat Aug 13 12:10:33 2022 daemon.info : 08[ENC] received unknown vendor ID: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
Sat Aug 13 12:10:33 2022 daemon.info : 08[ENC] received unknown vendor ID: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
Sat Aug 13 12:10:33 2022 daemon.info : 08[ENC] received unknown vendor ID: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
Sat Aug 13 12:10:33 2022 daemon.info : 08[IKE] 192.168.50.189 is initiating a Main Mode IKE_SA
Sat Aug 13 12:10:33 2022 authpriv.info : 08[IKE] 192.168.50.189 is initiating a Main Mode IKE_SA
Sat Aug 13 12:10:33 2022 daemon.info : 08[CFG] selected proposal: IKE:3DES_CBC/HMAC_SHA1_96/PRF_HMAC_SHA1/MODP_1024
Sat Aug 13 12:10:33 2022 daemon.info : 08[ENC] generating ID_PROT response 0 [ SA V V V V ]
Sat Aug 13 12:10:33 2022 daemon.info : 08[NET] sending packet: from 192.168.50.1[500] to 192.168.50.189[500] (156 bytes)
Sat Aug 13 12:10:33 2022 daemon.info : 12[NET] received packet: from 192.168.50.189[500] to 192.168.50.1[500] (260 bytes)
Sat Aug 13 12:10:33 2022 daemon.info : 12[ENC] parsed ID_PROT request 0 [ KE No NAT-D NAT-D ]
Sat Aug 13 12:10:33 2022 daemon.info : 12[IKE] faking NAT situation to enforce UDP encapsulation
Sat Aug 13 12:10:33 2022 daemon.info : 12[IKE] ENCRYPTION_ALGORITHM 3DES_CBC (key size 0) not supported!
Sat Aug 13 12:10:33 2022 daemon.info : 12[IKE] key derivation for pre-shared key failed
Sat Aug 13 12:10:33 2022 daemon.info : 12[ENC] generating INFORMATIONAL_V1 request 2568091693 [ HASH N(INVAL_KE) ]
Sat Aug 13 12:10:33 2022 daemon.info : 12[NET] sending packet: from 192.168.50.1[500] to 192.168.50.189[500] (80 bytes)

xcf13363175, 2022-08-29:
换softether

coolsnowwolf, 2026-09-06:
已核对日志：Windows 客户端协商到 `3DES_CBC`，但 `luci-app-ipsec-vpnd` 原依赖的 `strongswan-minimal` 未安装 `strongswan-mod-des`，导致后续报 `ENCRYPTION_ALGORITHM 3DES_CBC ... not supported`。已在 Makefile 增加 `+strongswan-mod-des` 依赖，保留现有 3DES proposal。修复提交 `1203cb4d2673a614dad249364473d9dcdab9b1c1`，已推送到 `openwrt-23.05`。

closed 2026-09-06 by coolsnowwolf