coolsnowwolf/luci #151: luci-theme-argon-mod 与 luci-theme-argon 主题冲突

smallprogram, 2022-08-17:
```shell
ConfiguriCollected errors:
 * check_data_file_clashes: Package luci-theme-argon-mod wants to install file /home/zhusir/lede_x64.config/build_dir/target-x86_64_musl/root-x86/usr/lib/lua/luci/view/themes/argon/footer.htm
        But that file is already provided by package  * luci-theme-argon
 * check_data_file_clashes: Package luci-theme-argon-mod wants to install file /home/zhusir/lede_x64.config/build_dir/target-x86_64_musl/root-x86/usr/lib/lua/luci/view/themes/argon/header.htm
        But that file is already provided by package  * luci-theme-argon
 * check_data_file_clashes: Package luci-theme-argon-mod wants to install file /home/zhusir/lede_x64.config/build_dir/target-x86_64_musl/root-x86/www/luci-static/argon/favicon.ico
        But that file is already provided by package  * luci-theme-argon
 * check_data_file_clashes: Package luci-theme-argon-mod wants to install file /home/zhusir/lede_x64.config/build_dir/target-x86_64_musl/root-x86/www/luci-static/argon/js/jquery.min.js
        But that file is already provided by package  * luci-theme-argon
 * check_data_file_clashes: Package luci-theme-argon-mod wants to install file /home/zhusir/lede_x64.config/build_dir/target-x86_64_musl/root-x86/www/luci-static/argon/js/script.js
        But that file is already provided by package  * luci-theme-argon
 * opkg_install_cmd: Cannot install package luci-theme-argon-mod.
```

aiamadeus, 2022-08-18:
这是肯定会冲突的，你只能选其中一个。

closed 2022-08-18 by aiamadeus

GreatMichaelLee, 2022-08-21:
> 这是肯定会冲突的，你只能选其中一个。

@aiamadeus 你自己试过mod编译没有，根本就编译不过！menu configure里，luci-app-argon是强制选择的，即便我手工编辑.config里吧luci-app-argon=y改成N,编译还是会报错！

[image]

EnnawYang, 2022-08-21:
> > 这是肯定会冲突的，你只能选其中一个。
> 
> @aiamadeus 你自己试过mod编译没有，根本就编译不过！menu configure里，luci-app-argon是强制选择的，即便我手工编辑.config里吧luci-app-argon=y改成N, 然后rm -rf ./tmp && make clean,进行干净编译还是会报错！
> 

那可能我用的是假的？

[image]

GreatMichaelLee, 2022-08-21:
> > > 这是肯定会冲突的，你只能选其中一个。
> > 
> > 
> > @aiamadeus 你自己试过mod编译没有，根本就编译不过！menu configure里，luci-app-argon是强制选择的，即便我手工编辑.config里吧luci-app-argon=y改成N, 然后rm -rf ./tmp && make clean,进行干净编译还是会报错！
> 
> 那可能我用的是假的？
> 
> [image]

you are developer, so you tell me.

aiamadeus, 2022-08-21:
> 你自己试过mod编译没有，根本就编译不过！

你自己没改 luci-app-argon-config 的依赖罢了。
另外说话麻烦客气一点，我不欠谁什么。

GreatMichaelLee, 2022-08-21:
![image](https://user-images.githubusercontent.com/20067401/185796034-4c86ed44-4694-4e89-b05b-bb1dbbd92853.png

GreatMichaelLee, 2022-08-21:
[image]

GreatMichaelLee, 2022-08-21:
> > 你自己试过mod编译没有，根本就编译不过！
> 
> 你自己没改 luci-app-argon-config 的依赖罢了。 另外说话麻烦客气一点，我不欠谁什么。

跟luci-app-argon-config有什么关系？我看menu configure里这个luci-theme-argon是被luci依赖了。我不知道为什么

GreatMichaelLee, 2022-08-21:
我搞定了，感谢两位，无意冒犯，经济不好，比较焦虑而已-_-!

smallprogram, 2022-08-21:
@GreatMichaelLee 哈哈哈哈😂