coolsnowwolf/luci #157: Adblock Plus 更新数据库后直接上不了网

mitsukileung, 2022-09-01:
开启Adblock Plus 并更新 “Adblock Plus 规则数据库”后，无法上网
记行之前更新后大概有6千多条记录，刚刚更新后有8万多条记录

xydche, 2022-09-01:
同样的问题，报如下日志
`Thu Sep  1 15:18:49 2022 daemon.crit dnsmasq[6508]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Thu Sep  1 15:18:49 2022 daemon.crit dnsmasq[6508]: FAILED to start up
Thu Sep  1 15:18:54 2022 daemon.crit dnsmasq[7479]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Thu Sep  1 15:18:54 2022 daemon.crit dnsmasq[7479]: FAILED to start up
Thu Sep  1 15:18:59 2022 daemon.crit dnsmasq[7502]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Thu Sep  1 15:18:59 2022 daemon.crit dnsmasq[7502]: FAILED to start up
Thu Sep  1 15:19:04 2022 daemon.crit dnsmasq[7523]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Thu Sep  1 15:19:04 2022 daemon.crit dnsmasq[7523]: FAILED to start up
Thu Sep  1 15:19:09 2022 daemon.crit dnsmasq[7534]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Thu Sep  1 15:19:09 2022 daemon.crit dnsmasq[7534]: FAILED to start up
Thu Sep  1 15:19:14 2022 daemon.crit dnsmasq[7546]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Thu Sep  1 15:19:14 2022 daemon.crit dnsmasq[7546]: FAILED to start up
Thu Sep  1 15:19:14 2022 daemon.info procd: Instance dnsmasq::cfg01411c s in a crash loop 6 crashes, 0 seconds since last crash`

直接导致无法解析DNS

xydche, 2022-09-01:
[image]

1054009064, 2022-09-04:
https://github.com/coolsnowwolf/luci/blob/39962e7f02b4ae085d48f6374f8da9400ed82f29/applications/luci-app-adbyby-plus/luasrc/controller/adbyby.lua#L57

找个精简的订阅链接的替换掉就好了，不知道这个为什么突然合并多了 6w 多条

closed 2022-09-09 by coolsnowwolf