coolsnowwolf/luci #160: 为什么luci-theme-argon不能升级？

cxgreat2014, 2022-10-01:
看到这个commit 撤销了该主题升级，不知道背后的原因

af5f733bb00dd9262a03a5ca0ba2c9cf50d33c3a

aiamadeus, 2022-10-10:
你好好看清楚，是已经有新的 argon 了，保留旧的又没什么问题。

closed 2022-10-10 by aiamadeus

cxgreat2014, 2022-10-10:
@aiamadeus 您好，首先感谢您的回复，个人有些疑惑，所以向大家请教。

从架构上而言，本 repo 属于 build source / 软件仓库 这样的单位，理应在一定频率上保持软件源码的稳定更新，但事实情况却与之相反，在更新完成之后反而进行了撤销，这一行为既不符合软件仓库的更新原则，也不符合逻辑常理。

如果说是旧版本的版本管理问题，git仓库已经做到了这一点，为什么还需要撤销更新呢？

aiamadeus, 2022-10-10:
你就不能好好看一下代码是吧，新的 argon 已经有了，那为啥要有两个一模一样的 argon？

cxgreat2014, 2022-10-10:
> 你就不能好好看一下代码是吧，新的 argon 已经有了，那为啥要有两个一模一样的 argon？

我将代码下载到本地进行了搜索，但依然没有找到新的 argon，不知道能提供一下链接吗？

EnnawYang, 2022-10-10:
[image]
嗯？不能翻一下吗？

cxgreat2014, 2022-10-10:
> [image]
> 
> 嗯？不能翻一下吗？

感谢您的回复，实际上我不太理解“新的 argon 已经有了”是什么意思，直观的理解是说仓库里已经有了新的argon，但我找到的和您截图的是一样的旧的argon。

回到主题，撤销主题所述提交是因为 升级版 / 修改版 的问题吗？亦或者背后有其他软件不兼容的 story ？

aiamadeus, 2022-10-11:
你是看不到有两个 argon 在哪吗？