coolsnowwolf/luci #161: luci-app-easymesh 的一些问题

VergilGao, 2022-10-31:
首先我要声明，我对lede项目并不是很熟悉，所以可能会有一些误解，欢迎指正。
openwrt在21.02采用了一个非常激进的改进，就是将原本的openssl迁移到了wolfssl，造成了非常多的问题。因为很多旧版本的软件包依然依赖openssl，而默认的配置中大多数都已经改为了默认依赖wolfssl，在编译时，openssl和wolfssl同时安装会产生冲突，所以我们会看到 https://github.com/ntlf9t/luci-app-easymesh/issues/7 这样的issue
我个人使用时更偏向wolfssl，所以我是硬编码修改了依赖到wolfssl，而官方库更多的是提供一个编译选项，可以手动选择openssl还是wolfssl
这是第一个问题。

第二个问题其实并不是luci-app-easymesh 本身的问题
因为一直以来luci并不支持 batman-adv 协议，所以相对应的网络端口的显示的是不支持的协议，前几天我在完善自用的路由器固件时发现openwrt其实已经在master分支引入了 luci-proto-batman-adv https://github.com/openwrt/luci/tree/master/protocols/luci-proto-batman-adv 试用之后发现这个网络端口的显示应该是暴露出来了 luci-app-easymesh 本身的一些配置项，因为对其中的配置不是很懂，所以我没多浏览，主要的疑问是 bat0 这样的端口，它默认是没有防火墙区域的，是否要对它配置防火墙？还有就是如果我开启了访客wifi并且设置了vlan，在mesh中又要如何配置呢？

VergilGao, 2022-11-07:
luci-app-easymesh用了一周，正常跑都没问题，一旦重启某个节点就会出现各种奇怪的bug

于是这几天尝试用openwrt官方推荐的方式配置mesh
我删除了luci-app-easymesh
安装了openwrt官方推荐的那几个包
安装mesh11sd，保持默认
手动添加 bat 网络端口
手动添加 mesh0 无线网络，在luci界面里新建了 mesh0 interface
设置 mesh0 interface 为 batadv_hardif
编辑 /etc/config/network ，将 mesh0 interface 的 option type 'bridge' 删除
没有重启的情况下，目前来说一切正常

现在的想法是，把这个过程写个luci插件，这样方便配置，还能监控节点的状态

zxlhhyccc, 2022-11-24:
> luci-app-easymesh用了一周，正常跑都没问题，一旦重启某个节点就会出现各种奇怪的bug
>
> 于是这几天尝试用openwrt官方推荐的方式配置mesh 我删除了luci-app-easymesh 安装了openwrt官方推荐的那几个包 安装mesh11sd，保持默认 手动添加 bat 网络端口 手动添加 mesh0 无线网络，在luci界面里新建了 mesh0 interface [...]

需要安装官方哪几个包？我看看能不能写一个。。。

VergilGao, 2022-11-26:
> 

我这段时间很忙，所以一直没有再弄这个事情，现在的问题时根据官方的方案，运行一段时间后会出现接入点繁忙的错误，然后wifi就连不上了
看日志也看不出什么来
需要的包有
- luci-proto-batman-adv 这个能在接口节点看到具体的batman协议流量
- wpad-wolfssl 或者wpad-openssl 二选一
- mesh11sd 这个是核心，负责底层

如果有需要测试的话可以找我，我尽量安排时间来测……

zxlhhyccc, 2022-11-26:
> 我是看的这两个教程来部署的： https://openwrt.org/docs/guide-user/network/wifi/mesh/mesh11sd https://openwrt.org/docs/guide-user/network/wifi/mesh/batman
> 
> 我这段时间很忙，所以一直没有再弄这个事情，现在的问题时根据官方的方案，运行一段时间后会出现接入点繁忙的错误，然后wifi就连不上了 看日志也看不出什么来 需要的包有
> 
> * luci-proto-batman-adv 这个能在接口节点看到具体的batman协议流量
> * wpad-wolfssl 或者wpad-openssl 二选一
> * mesh11sd 这个是核心，负责底层
> 
> 如果有需要测试的话可以找我，我尽量安排时间来测……

官方master分支有这些包可以编译，你的意思是搞luci吗？

VergilGao, 2022-11-27:
我的配置应该是有问题的，暂时还不知道问题在哪，我今天正好有时间，先在家简单测试一下

coolsnowwolf, 2026-09-06:
经核对，`luci-app-easymesh` 已在同步官方 `openwrt-23.05` 时移除（提交 `3f0c510ed4dcb4c1ae12d0e07bcb5e92dca50a7e`），当前仓库不存在该应用或可维护代码。因此无法在本仓库继续处理旧版 OpenSSL/wolfSSL 依赖及 batman-adv 配置咨询，标记为 not planned。

closed 2026-09-06 by coolsnowwolf