coolsnowwolf/luci #165: TurboACC高性能模式，貌似无法启动？

xiangfeidexiaohuo, 2022-11-22:
拉取了最新源码编译，选了高性能模式，还是显示未运行，选兼容模式，确实是运行的。

选禁用和高性能模式：
Chain zone_wan_postrouting (0 references)
target prot opt source destination
postrouting_wan_rule all -- anywhere anywhere /* !fw3: Custom wan postrouting rule chain /
MASQUERADE all -- anywhere anywhere / !fw3 */

[image]

选兼容模式：
Chain zone_wan_postrouting (0 references)
target prot opt source destination
postrouting_wan_rule all -- anywhere anywhere /* !fw3: Custom wan postrouting rule chain /
FULLCONENAT all -- anywhere anywhere / !fw3 */

[image]

zxlhhyccc, 2022-11-22:
我也是一样无法启动。。。。

aiamadeus, 2022-11-23:
很简单，就只有 luci 支持了而已，高性能模式不存在。

closed 2022-11-23 by aiamadeus

zxlhhyccc, 2022-11-23:
> 很简单，就只有 luci 支持了而已，高性能模式不存在。

有道理，但为啥要添加高性能？

WCJ23333, 2022-11-24:
在防火墙中更改为高性能模式，turbo acc中状态无法同步，仍然显示兼容模式
查看/etc/config/turboacc  发现  option fullcone_nat '1'  也就是值没有更改为2

更改为高性能模式时，全锥形NAT会显示未运行