coolsnowwolf/luci #179: 解决 luci-app-baidupcs-web 在官方 OpenWrt 22.03.2 中无法运行的问题

zhengmz, 2022-12-28:
需要修改两个地方：

1. 修改 `/etc/init.d/baidupcs-web`: 将此行注释，在行首加 `#`
https://github.com/coolsnowwolf/luci/blob/a7f754dfbe07faa12bb04746a6ec751a94d13d61/applications/luci-app-baidupcs-web/root/etc/init.d/baidupcs-web#L15
修改成: `#	stop`

2. 修改 `view/baidupcs-web/baidupcs-web_status.htm`: 在 id 中增加 `widget.`
https://github.com/coolsnowwolf/luci/blob/a7f754dfbe07faa12bb04746a6ec751a94d13d61/applications/luci-app-baidupcs-web/luasrc/view/baidupcs-web/baidupcs-web_status.htm#L27
修改成： `	var clientPort = window.document.getElementById("widget.cbid.baidupcs-web.config.port").value`

hue715, 2023-01-21:
这是因为/etc/init.d/baidupcs-web与执行文件重名了，个人觉得可以把luci界面更名

> 需要修改两个地方：
> 
> 1. 修改 `/etc/init.d/baidupcs-web`: 将此行注释，在行首加 `#`
>    https://github.com/coolsnowwolf/luci/blob/a7f754dfbe07faa12bb04746a6ec751a94d13d61/applications/luci-app-baidupcs-web/root/etc/init.d/baidupcs-web#L15
>    
>    修改成: `#	stop`

zhengmz, 2023-02-01:
> 这是因为/etc/init.d/baidupcs-web与执行文件重名了，个人觉得可以效仿此更改#101
> 
> > 需要修改两个地方：
> > 
> > 1. 修改 `/etc/init.d/baidupcs-web`: 将此行注释，在行首加 `#`
> >    https://github.com/coolsnowwolf/luci/blob/a7f754dfbe07faa12bb04746a6ec751a94d13d61/applications/luci-app-baidupcs-web/root/etc/init.d/baidupcs-web#L15
> >    
> >    修改成: `#	stop`

有理

PR #189 (2023-02-09): luci-app-baidupcs-web: Fix startup error
Fixed the problem that the service failed to start  (Fix #179)
closed 2023-02-09 by Beginner-Go