coolsnowwolf/luci #207: luci-app-zerotier 无法在官方 22.03+ 上使用 nat 功能

zhengmz, 2023-03-26:
我看一下，应该是下面这几条语句

https://github.com/coolsnowwolf/luci/blob/adda8b2ed340d1be5429393fb4a788a08c128911/applications/luci-app-zerotier/root/etc/zerotier.start#L22-L26

不知道哪个大佬可以翻译成 nftables，谢谢

我尝试安装以下两种方式都无法成功：

1. 安装 iptables-nft，报 '表 nat' 不存在
2. 将 iptables 改为 iptables-legacy，报 '表 nat' 不存在

TianWeiShuiChun, 2023-03-29:
不确定对不对。
[image]

请使用以下命令将iptables规则转换为nftables规则：

nft add rule filter FORWARD iif "$i" accept
nft add rule filter FORWARD oif "$i" accept
nft add rule nat POSTROUTING oif "$i" masquerade
ip_segment="$(ip route | grep "dev $i proto kernel" | awk '{print $1}')"
nft add rule nat POSTROUTING ip saddr "${ip_segment}" masquerade
这些规则将在nftables中添加与iptables中相同的规则。请注意，nftables的语法与iptables不同，因此需要进行一些更改。

xieqiyou, 2023-05-14:
我是不用这个按钮来实现这个功能的
1.运行zerotier连上
2.
在网络——————防火墙下面新建一个新的防火墙
注意勾选 IP 动态伪装
[image]
3.
在网络接口下面新建一个新的接口
[image]

4.把这个接口防火墙配置到新建的那个防火墙
[image]

zhengmz, 2023-05-19:
好的，谢谢，我试一下

zhengmz, 2023-05-29:
@xieqiyou 我提交一下合并请求 #230 ，通过脚本实现 nat 功能

适合定制固件使用，不需要人工设置

closed 2023-05-30 by zhengmz