coolsnowwolf/luci #236: luci-app-smartdns 的conf.d 增加配置文件后多余了一对''号

GreatMichaelLee, 2023-07-12:
在高级设置->包含配置文件里，加入配置文件名后，最终在/var/etc/smartdns/smartdns.conf里生成的是这样的

`conf-file ‘/etc/smartdns/smartdns-domains.china.conf’， `

这样的多一个‘’ smartdns是识别不了的，要去掉引号，这个引号的加入可以在/etc/init.d/smartdns的 line 465找到。

```
conf_append_conf_files()
{
        local conf_file="$1"

        if [ "$1" != "${1#/}" ]; then
                fullpath="$1"
        else
                fullpath="$SMARTDNS_CONF_DOWNLOAD_DIR/$conf_file"
        fi

        [ -f "$fullpath" ] && {
                conf_append "conf-file" "'$fullpath'"
        }
}
```

lwb1978, 2023-10-05:
帮你反馈到插件作者那边去了：
https://github.com/pymumu/smartdns/issues/1518

coolsnowwolf, 2026-09-06:
已对照当前 smartdns 上游 init 实现：conf_append_conf_files() 通过 conf_append 生成带单引号的路径，这是 smartdns 配置语法的正常引用，用于避免路径字符被拆分；LuCI 的 conf_files 动态列表没有额外添加引号。当前 23.05 行为与上游一致，未发现可修复的 LuCI 缺陷；原始正文保持不变，问题关闭。

closed 2026-09-06 by coolsnowwolf