coolsnowwolf/luci #273: docker的配置文件daemon.json为什么在/tmp/dockerd文件夹里面啊？

Pacio, 2024-03-20:
正常不是应该在/etc/docker下的吗？
然而/etc/docker根本不存在

qingtian110, 2024-03-27:
因为正如你所看见，它是 temp file，它会在 init.d 重启操作时重新生成，所以它没有在 /etc 的必要，它也不允许自定义修改。

https://github.com/openwrt/packages/blob/800d69b4b4efb6ca816bde4945184eaf5ce25328/utils/dockerd/files/dockerd.init#L208

Pacio, 2024-03-27:
> 因为正如你所看见，它是 temp file，它会在 init.d 重启操作时重新生成，所以它没有在 /etc 的必要，它也不允许自定义修改。
> 
> https://github.com/openwrt/packages/blob/800d69b4b4efb6ca816bde4945184eaf5ce25328/utils/dockerd/files/dockerd.init#L208

我手动创建了/etc/docker/daemon.json，自定义配置可以生效

qingtian110, 2024-03-27:
好吧，那你是蛮厉害的了，dockerd 默认的配置文件就是 `/etc/docker/daemon.json`

但是 openwrt 的 dockerd init.d 只有在生成配置文件失败时才会不使用自定义配置启动守护进程。

```sh
	if [ -z "${DOCKERD_CONF}" ]; then
		procd_set_param command /usr/bin/dockerd
	else
		procd_set_param command /usr/bin/dockerd --config-file="${DOCKERD_CONF}"
	fi
```

https://github.com/openwrt/packages/blob/800d69b4b4efb6ca816bde4945184eaf5ce25328/utils/dockerd/files/dockerd.init#L221

你能在 luci 的 openwrt 下让它不生成配置，确实是大佬级

closed 2024-04-05 by Pacio