coolsnowwolf/luci #279: 进程菜单不显示任何数据

smallprogram, 2024-05-04:
[image]

尝试SSH链接到OP，使用PS命令可以正常显示进程信息

[image]

请问各位大佬，该如何修复该问题

TheLastManT1, 2024-05-22:
你的busybox有问题，可能缺少-w参数。如图，左边是正常的，右边是不正常的。
[image]
我用apk安装软件包的时候就遇见过，参考博客
https://blog.revincx.icu/posts/never-use-apk-on-openwrt#%E5%B9%95%E5%90%8E%E7%9C%9F%E5%87%B6%E7%9A%84%E6%B5%AE%E7%8E%B0

smallprogram, 2024-05-22:
> 你的busybox有问题，可能缺少-w参数。如图，左边是正常的，右边是不正常的。 [...]

好的，感谢回复，我试试

smallprogram, 2024-05-22:
@TheLastManT1 
[image]
我的-w参数也是没问题的

closed 2024-11-17 by coolsnowwolf