coolsnowwolf/luci #282: luci-app-acme申请cloudfare证书失败

yflwz, 2024-06-05:
按要求已经配置好相关的参数
[image]
[image]
主要报错是以下这段
invalid domain
Error add txt for domain:_acme-challenge.gdzglc.com
_on_issue_err
Please add '--debug' or '--log' to check more details.
See: https://github.com/acmesh-official/acme.sh/wiki/How-to-debug-acme.sh
下面附上完整的日志
[acme.txt](https://github.com/user-attachments/files/15575580/acme.txt)

yflwz, 2024-06-05:
系统版本
[image]

yflwz, 2024-06-05:
用手动的方式敲命令可以申请，只是不能全自动比较麻烦，申请之后要重新在uhttpd页面上传新的证书。

SuitThug3977, 2024-12-30:
acme-deploy
acme-dnsapi
acme-notify
这3包装了吗？

closed 2026-09-06 by coolsnowwolf