coolsnowwolf/luci #320: 用luci的openwrt-23.05分支版本编译完固件发现Netdata插件菜单找不到，删掉netdata软件包后Dockerman菜单就都显示全了。

SuitThug3977, 2024-12-30:
请赐教，不知道到底啥问题，用https://github.com/sirpdboy/luci-app-netdata的源码直接编译的也不行，还直接影响了Dockerman的运行。

[image]
↑NETDATA（实时监控）插件菜单找不到

[image]
↑Dockerman菜单不全

[image]
↑卸载netdata，虽然报错，但是执行完成后软件包都不在列表里了。

[image]
↑神奇的是卸载netdata后，Dockerman菜单居然就都显示全了！

[netdata_log.txt](https://github.com/user-attachments/files/18272385/netdata_log.txt)
↑Netdata启动日志

hztbhtbh, 2025-01-19:
你试试编译的时候先选上openclash然后再看看有没有netdata.这两个插件互相依赖应该是。

coolsnowwolf, 2025-03-15:
均已解决

closed 2025-04-11 by coolsnowwolf